smlight-dev/slzb-06-firmware #9: Pairing problem on some Legrand devices

samohthou, 2023-01-15:
Hello,
I bought the SMLIGHT - SLZB 06 adapter to replace my Conbee II in order to have a better reach.
I use Zigbee2mqtt on Home assistant.
I updated the Coordinator firmware and ESP32 firmware with the latest versions:

```
Model: | SLZB-06
-- | --
ESP32 Firmware ver: | 0.9.8 (Jan 2 2023)
ESP32 version: | ESP32-D0WDQ5
ESP32 temperature: | 40.22 °C
ESP32 frequency: | 2 cores @ 240 MHz
ESP32 flash size: | 16 Mb, external
ESP32 Free heap: | 231 / 303 KiB
CC2652P version: | CC2652PP1FRGZR
CC2652P frequency: | 48 MHz
CC2652P flash size: | 352 Kb
```

I had no problem to configure Zigbee2mqtt in tcp on the SLZB 06.
No problem either to pair my aquara door sensors and my philips hue presence detectors.

On the other hand, impossible to pair my Legrand ZLGP17 switch although compatible with Zigbee2Mqtt: https://www.zigbee2mqtt.io/devices/ZLGP17_ZLGP18.html
I have no trace of a pairing attempt in the zigbee2mqtt logs
However, I have not had any problems with the pairing of these switches on the Conbee II.

Do you know why these switches are not recognized and do you know if it will be possible to make it visible later?
Thanks in advance.

sergge1, 2023-01-16:
_I have no trace of a pairing attempt in the zigbee2mqtt logs_
did you follow the same pairing steps as you did with Conbee II (e.g. `hold the configuration button (on the back of the device) and press the button (up or down) on the first side, then do the same for the second side.`? If you do not see the pairing attempt - it means the end device does not start pairing. I wanted to recommend opening a question in z2m, but I realized that the connection does not start at all.

samohthou, 2023-01-16:
Hello,
Thank you for your feedback, but indeed it seems that the problem is at the level of the SLZB 06 because to test I have:
- Delete the peripherals on the conbee II
- Put back a base and conf by default in Z2M
- Connected in tcp the SLZB 06
- Tried to pair my switch by following the steps of pressing the back button and the front button without success on the SLZB 06.
- Rollback Z2M by connecting my conbee II
- Pair my switch which is added directly on the conbee II
This makes me say that the problem is more likely to come from the SLZB 06 than Z2M.

sergge1, 2023-01-16:
The hardware itself is just an intermediary in communication so it is not the case taking into account that pairing of other devices is Ok. If you even do not receive a pairing attempt means that the problem is somewhere else.

Quick Googling provides a lot of discussions on that topic, like [here](https://github.com/Koenkk/Z-Stack-firmware/issues/318#issuecomment-964886178). As far as I understood your device is communicating not by Zigbee but with Zigee Green Power (ZGP) protocol. While z2m itself can support the device, z-stack firmware can commit it. I did not understand whether the support has been added, you can try a dev Z-Stack firmware from dev branch to try

samohthou, 2023-01-19:
Thanks for your feedback, I tried to use DEV firmware from Z-Stack-firmware. Once the update is done, impossible to reconnect SLZB-06 on Z2M ... 
So I'm going to rollback on my conbee II while waiting for a future firmware that maybe solve my problem.

sergge1, 2023-01-23:
Got it. Thank you for your feedback.

closed 2023-01-23 by sergge1