smlight-dev/slzb-06-firmware #10: Blank web interface after upgrading to v0.9.8

damian-w, 2023-01-17:
After flashing and updating the ESP32 [over-the-air](https://smlight.tech/manual/slzb-06/guide/flashing-and-updating/updating-esp32.html#updating-esp32-over-the-air-distantly), the web interface is now blank. I have rebooted the SLZB-06 and also deleted my browser cache without any luck. Any ideas would be appreciated!

[Screenshot](https://imgur.com/a/fuW4T0A)

sergge1, 2023-01-17:
> Any ideas would be appreciated!

For support - it is better to submit the question to the support ticketing system on the Manual page (follow QR from the package). 

Briefly, did you refresh the web page in your browser?
If this does not help, can you please try to flash over the USB? Flashing over the USB erases all the information at ESP32 and flash it again; here is the manual [manual](https://smlight.tech/manual/slzb-06/guide/flashing-and-updating/updating-esp32.html#updating-esp32-via-usb)

damian-w, 2023-01-17:
Full reboot & wipe fixed this issue. Closing.

closed 2023-01-17 by damian-w