smlight-dev/slzb-06-firmware #12: Unable to connect zigbee2mqtt to SLZB-06 via ethernet and USB

beol, 2023-01-19:
I've recently purchased SLZB-6 from AliExpress. I've been unable to connect it to my z2m setup via USB or ethernet with:
`Zigbee2MQTT:error 2023-01-19 18:23:31: Error while starting zigbee-herdsman                                                          
Zigbee2MQTT:error 2023-01-19 18:23:31: Failed to start zigbee
Zigbee2MQTT:error 2023-01-19 18:23:31: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-19 18:23:31: Exiting...
Zigbee2MQTT:error 2023-01-19 18:23:32: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)               
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)       
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)                          
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:27)                                                                  
    at start (/opt/zigbee2mqtt/index.js:107:5)
`

I've updated the firmware and also the reflash the coordinator firmware. I'm not sure if the unit is working or broken.

sergge1, 2023-01-19:
hi, please submit your request through the [ticketing system](https://smlight.tech/support/) so support will contact you and will figure out your problem.

beol, 2023-01-19:
I've reached out here because I can't tell whether my submission to the ticketing system was received or not, there has been no acknowledgement of ticket being created.

senz, 2023-01-23:
Hello. same problem here.
Also had submitted a request with your system, but no answer.
This repo is about a firmware for a specific device (its even in the name of the repo), so lets discuss this problem here, for benefit of the other people facing exact same issue.
What info do you need for debugging @sergge1 ?

sergge1, 2023-01-23:
Dear visitors, I am maintaining ESP firmware here. For inquiries about the device itself (everything that does not relate to the ESP firmware) - please contact them via ticketing or email. Thank you!

closed 2023-01-23 by sergge1