smlight-dev/slzb-06-firmware #16: After updating ESP32 firmware to 0.9.8, version remains as 0.9.1

jalapenopuzzle, 2023-02-15:
AFTER following the firmware upgrade procedure with firmware file
0.9.8.20230201.bin

1) Download the firmware from https://github.com/smlight-dev/slzb-06-firmware/releases/tag/v0.9.8
2) Connect to my SLZB-06 by ethernet
3) Navigate to the "System and Tools" page
4) Click "Choose file" and choose the file 0.9.8.20230201.bin (1,042,784 bytes)
5) Click "ESP32 OTA Update"
6) "progress 100%" is displayed
7) Click ESP32 Restart
8) Visit the device and pull the ethernet cable, press the reset button, re-connect the ethernet cable, wait for lights to flash, release reset button (procedure recommended by firmware update notes)
9) Navigate to the "Status" page
10) The following information is displayed:
Model: | SLZB-06
ESP32 Firmware ver: | 0.9.1
ESP32 version: | ESP32-D0WDQ5

I expected to see firmware version 0.9.8 displayed.

I don't know if the firmware update has gone wrong, or the firmware contains the wrong version number for display.

sergge1, 2023-02-15:
Firmware number contains the right version, double-checked. 
Can you please try to do it via USB as per manual?
Otherwise, after step (6) can you please wait ~1 minute  before (7) restart?

Tarik2142, 2023-02-27:
This can happen if several packets missed during the update (perhaps not a stable network).
In this case, the web interface will show 100%, but the new firmware will not be accepted because it is damaged and the device will run the previous firmware.
Just start update again.

jalapenopuzzle, 2023-02-27:
I attempted the upgrade about 5 times.
The two SLZB-06 are connected directly to a UniFi USW-Pro-48-PoE, which is 
also directly connected to the Windows 10 laptop doing the firmware upgrade.

I'm not expecting dropped packets on this network, but I haven't proven 
that there aren't any dropped packets.

Ben.

Tarik2142, 2023-02-27:
@jalapenopuzzle 
Do you see text "Update completed" after 100%?
[image]

sergge1, 2023-02-28:
@jalapenopuzzle  can you try to update via USB?https://smlight.tech/manual/slzb-06/guide/flashing-and-updating/updating-esp32.html#updating-esp32-via-usb

closed 2023-04-13 by sergge1