smlight-dev/slzb-06-firmware #17: SLZB-06 Doesn't realize it's connected to LAN, turns on AP

shlomki, 2023-02-24:
Hi, I've just installed a new SLZB-06.
It has the latest firmware of 0.9.8.
I've connected it to LAN, it got an IP address as expected, and I can access the web interface.

However, for some reason it doesn't notice that it's already connected to LAN, and it turns on the AP.
I would like to torn the AP off but I can't find where to do it.
Please help understand the problem. 

[image]

[image]

[image]

sergge1, 2023-02-25:
Hi, @shlomki  thanks for letting me know.
Most probably, pings in your LAN network are disabled by your switch/router settings.
Could you please go to the "Config Ethernet" tab and check the box "Disable gateway ping" (should be Disabled).
This will solve your issue.
Just FYI, that is because the system makes ping check and if does not receive an answer, it treats that the ethernet is not properly connected and starts AP.

sergge1, 2023-02-25:
we will disable "ping check" by default in the next release

shlomki, 2023-02-25:
Thanks @sergge1. I believe you're right. I'll be able to verify this tomorrow. 

I would suggest to improve the UX as well: move the "Disable gateway ping" to the top of the page, or to another place. Currently it seems like it's a part of the DHCP config. I thought that if DHCP is on I can't control this toggle, because it's right under a few disabled text boxes. 
Now, thanks to your comment, I understand that I can. Misunderstanding on my part, but it seems that this can be clearer with better placement of this toggle.

sergge1, 2023-02-25:
> I would suggest to improve the UX as well.
Thanks, fully agree. Will implement this comment in next release, I will inform as soon as it is ready :)

closed 2023-02-28 by sergge1