smlight-dev/slzb-06-firmware #19: Status shows Ethernet is not connected when it is

ZSasha0, 2023-03-13:
I'm using the device in wired etherned mode, however on the Status page the systems shows it as disconnected.
I do NOT use DHCP - device's IP is fixed.

Model:	SLZB-06
ESP32 Firmware ver:	0.9.8 (Jan 2 2023)
ESP32 version:	ESP32-D0WDQ5
ESP32 temperature:	63.44 °C
ESP32 frequency:	2 cores @ 240 MHz
ESP32 flash size:	16 Mb, external
ESP32 Free heap:	195 / 302 KiB
CC2652P version:	CC2652PP1FRGZR
CC2652P frequency:	48 MHz
CC2652P flash size:	352 Kb

Tarik2142, 2023-03-13:
Can you flash 0.9.9 dev build?
https://github.com/smlight-dev/slzb-06-firmware/releases/tag/v0.9.9-dev
I fixed this problem in 0.9.9

Tarik2142, 2023-03-16:
Fixed in 0.9.9 https://github.com/smlight-dev/slzb-06-firmware/releases/tag/v0.9.9

closed 2023-03-16 by Tarik2142