smlight-dev/slzb-06-firmware #21: Feature request: Prometheus / OpenMetrics metrics endpoint

junosuarez, 2023-03-31:
The prometheus / OpenMetrics metrics format is a simple, text-based way for Prometheus (and other monitoring tools) to collect status information over http - you can read more about it at https://scoutapm.com/blog/prometheus-metrics

The request is to add an http endpoint to the builtin server called `/metrics` that would return the same stats as the status page. This would make it easy to implement network monitoring, for example within a Grafana dashboard. 

Note, I am currently working on implementing the same in Zigbee2Mqtt, see https://github.com/Koenkk/zigbee2mqtt/issues/4521

Tarik2142, 2023-03-31:
unclear protocol, I don't understand what the server should return

junosuarez, 2023-03-31:
The server would return a text file with the metric names and current value. The spec documenting the various metric types is https://github.com/OpenObservability/OpenMetrics/blob/main/specification/OpenMetrics.md

For example, I found that the current status page is getting an api response with some json like
```json
{
  "VERSION": "0.9.8 (Jan  2 2023)",
  "connectedSocketStatus": "Yes, 1 connections",
  "connectedSocket": "0 d 00:08:41",
  "operationalMode": "Zigbee-to-Ethernet",
  "ethDhcp": "On",
  "connectedEther": "Yes",
  "ethConnection": "Connected",
  "ethSpd": "100 Mbps",
  "ethIp": "10.10.8.4",
  "etchMask": "255.255.128.0",
  "ethGate": "10.10.8.1",
  "uptime": "0 d 00:53:57",
  "deviceTemp": "40.78",
  "hwRev": "SLZB-06",
  "espModel": "ESP32-D0WDQ5",
  "espCores": "2",
  "espFreq": "240",
  "espHeapFree": "221",
  "espHeapSize": "303",
  "espFlashType": "external",
  "espFlashSize": "16",
  "wifiEnabled": "No",
  "wifiConnected": "Not connected",
  "wifiMode": "Off",
  "wifiSsid": "Off",
  "wifiIp": "Off",
  "wifiSubnet": "Off",
  "wifiGate": "Off",
  "wifiRssi": "Off",
  "wifiDhcp": "Off",
  "wifiModeAP": "No",
  "wifiModeAPStatus": "Not started"
}
```

Depending on what the metrics are named, this could be represented like:
```
connected_socket_status=1
connected_socket_connections=1
connected_socket_uptime_seconds=521
connected_ether_status=1
connected_ether_speed_mbps=100
connected_ether_uptime_seconds=3237
device_temp=40.78
...
```
I'm not sure what other metrics would be interesting to expose - the uart socket connection status is the most interesting one for my needs. 

## Alternatives considered
I could create a separate process to scrape the  data as-is and reformat it as OpenMetrics. This is probably what I'll end up doing in the short term. However, it's less than perfect because it introduces another point of failure (the converter process). Best practice for simplicity and reliability is for services or devices to report their own metrics data directly.

Tarik2142, 2023-03-31:
> connected_socket_status=1
connected_socket_connections=1
connected_socket_uptime_seconds=521
connected_ether_status=1
connected_ether_speed_mbps=100
connected_ether_uptime_seconds=3237
device_temp=40.78

Ok, now it's clear.

The device doesn't have many parameters that are updated often, so I think it makes sense to export only those parameters
[image]

Tarik2142, 2023-04-03:
@junosuarez check this build
[firmware.zip](https://github.com/smlight-dev/slzb-06-firmware/files/11141445/firmware.zip)
server endpoint **/metrics**
[image]

fatpat, 2023-04-07:
Hi,

definetly the way to go. OpenMetrics export is a must have. It opens the door to the observability world (prometheus/alertmanager/grafana ... Name it)

and using an external exporter must be considered as an alternative way because including the exports directly in the soft is the best practice to ensure to always have metrics following changes in the code base.

Tarik2142, 2023-04-19:
Hi, @junosuarez
Can you confirm that everything is right?

onedr0p, 2023-04-19:
@Tarik2142 From that screen shot the metrics do not appear to be in the correct format.

Have you had a look here?

https://prometheus.io/docs/instrumenting/writing_exporters/

onedr0p, 2023-05-18:
I didn't give the FW a try yet however, there should be a space in the output and it would be great if the metrics started with `smlight_` or some other unique identifier to help segregate the metrics from others that Prometheus consumes. e.g.

```
smlight_connected_socket_status{device="SLZB-06"}<space>1
```

onedr0p, 2023-05-18:
Device name is actually not needed since Prometheus will add an instance label (with the IP address of the controller) to the metrics already.

Tarik2142, 2023-05-22:
@onedr0p so is it done?
[firmware.zip](https://github.com/smlight-dev/slzb-06-firmware/files/11531338/firmware.zip)
[image]

onedr0p, 2023-05-22:
I'll update my firmware today and test.

onedr0p, 2023-05-22:
I put the zip here, correct?
[image]

Tarik2142, 2023-05-22:
First, unzip the archive, then click "Choose file" and start flashing

onedr0p, 2023-05-22:
Almost there.. it looks like there might be some formatting issues on that metrics endpoint

[image]

onedr0p, 2023-05-22:
https://stackoverflow.com/questions/65388098/how-to-parse-prometheus-data

Tarik2142, 2023-05-22:
@onedr0p yep, i use `\r\n` for line ending
check this fw [firmware.zip](https://github.com/smlight-dev/slzb-06-firmware/files/11531804/firmware.zip)

onedr0p, 2023-05-22:
That's working now, any plans to add more metrics?

Tarik2142, 2023-05-22:
@onedr0p what metrics do you need?

bdd, 2023-05-27:
@Tarik2142 are you planning to push either the feature branch for openmetrics format and endpoint support to public repository, or maybe even merge it to `dev` branch?

I have a few more PRs coming soon, and I'd love to make it easier for you to not fight with a plethora of merge conflicts.

Tarik2142, 2023-05-30:
@bdd Thanks for your PRs, I will try to update dev branch in the coming days so that you work with the current firmware.

seandonohoo, 2023-07-11:
Can also confirm the firmware above is working great with prometheus :)

xvisor16, 2023-07-27:
I can also confirm that it works! Sadly with this firmware the version of the zigbee firmware isn't visible anymore, also when trying to update zigbee it's says "Your current firmware revision: Unknown". I did manage to flash my coordinator firmware to the latest one. So that did work just fine.

onedr0p, 2023-07-28:
Are there links to download the firmware instead for beta builds?

xvisor16, 2023-07-28:
@Tarik2142, Thnx found it and flashed it. Working!
@onedr0p, You can find it on Github: https://github.com/smlight-dev/slzb-06-firmware/releases
But only use this if you are already running 1.0.0-dev!! Otherwise use the flasher as Tarik said!

onedr0p, 2023-07-28:
> @onedr0p yep, i use `\r\n` for line ending check this fw [firmware.zip](https://github.com/smlight-dev/slzb-06-firmware/files/11531804/firmware.zip)

I last updated to this, does that mean I can use the Github release artifacts to upgrade?

sergge1, 2023-08-17:
@onedr0p @xvisor16 is this done in the dev version? if yes, I will close. if no response within 24H I will close the issue as well.

onedr0p, 2023-08-17:
I'm still unsure how to update being on the dev version, can you answer my last question?

Tarik2142, 2023-08-17:
> I'm still unsure how to update being on the dev version, can you answer my last question?

If your current firmware version is less than 1.0, then firmware 1.0 or higher must be flashed via USB. If you have already flashed 1.0+ via USB, you can update via OTA to new firmware in the future.

Tarik2142, 2023-08-17:
> I think my confusion is around being on the dev version which I did not flash via USB, so if I understand right I can flash the newer firmwares from the web UI?
> 
> My current version: [image]

You can use it like this, but Zigbee OTA updates will not work until you flash the firmware via USB

onedr0p, 2023-08-17:
Ah that clears it up. I'll await for 1.0 to drop and flash it via USB.
Back to the issue here, I think this can be closed now... @junosuarez has appear to drop off the planet again:D

closed 2023-08-17 by sergge1

xvisor16, 2023-08-25:
@sergge1 sorry for the late reaction, but yes!