smlight-dev/slzb-06-firmware #22: WIFI access point is enabled

jwillmer, 2023-04-07:
How can I disable the access point? I only want `Zigbee to Ethernet` but I see that the device created a access point and can also see it in the status page:

[image]

jwillmer, 2023-04-07:
I figured it out. I did not set any `Subnet Mask` in the network config since the grayed one was already correct - and everything worked. Once I added the `Subnet Mask` manually the AP got disabled. There should be a validation for `Subnet Mask` if it is critical to set. And also a warning that AP is started since this is a security concern.

Tarik2142, 2023-04-08:
@jwillmer Thanks for this info, it really looks like a validation is missed. This will be fixed in 1.0.0

Tarik2142, 2023-04-12:
Fixed in https://github.com/smlight-dev/slzb-06-firmware/releases/tag/v1.0.0-dev

closed 2023-04-12 by Tarik2142