smlight-dev/slzb-06-firmware #27: [Manuals Update] Flash without GUI

Ramalama2, 2023-05-11:
Hi, everything is great so far about this Stick!
Thanks Guys for making such a great PCB that covers everything!

I have just one request,
My Stick is in a location where its a pain in the ass to connect an usb cable to (To flash via usb)
Lets say we have to flash via USB to update from 0.99 to 1.0.0

So basically i had to connect an raspberry pi to the SLZB-06 Stick, and ssh onto the raspberry, clone the espflasher repository and flash from my computer over ssh from the pi the slzb stick :-)

The raspberry here has of course no gui.

That way should be described in the Manual either.

Another sidenote is: I have an M2 Macbook Pro with MacOS Ventura, the cp2102x driver isn't working on Ventura, at least not for the ARM (Apple Silicon) Plattform, maybe it does on the old intel X86_64 Platform, but however, the driver isn't working so you cant flash with MacOS.

Cheers

Ramalama2, 2023-05-11:
I found another issue, it seems for me that an usb-c to usb-c connection doesn't work (i tryed with some laptops)
But usb-c to usb-a connection works perfectly fine.

Never seen such an issue, ive tryed with a definitively working c to c cable with what i flashed other esp's already.

sylvaindd, 2023-05-26:
> I found another issue, it seems for me that an usb-c to usb-c connection doesn't work (i tryed with some laptops) But usb-c to usb-a connection works perfectly fine.
> 
> Never seen such an issue, ive tryed with a definitively working c to c cable with what i flashed other esp's already.

Same here for me, usb c-c doesn't work, usb c-a OK.

Tarik2142, 2023-07-10:
> I have again and again this error: [image]
> 
> Tested USB A-C cable, resetting, holding and nothing :)

Try PC witch Intel chipset
OR
Short this pins for download mode
[image]

S474N, 2023-07-10:
I have Intel chipset. But now have also problem with connecting to USB, there isn't visible device and no "windows sound" of connecting. After restart PC the same (also tested with another USB port). Must be also connected to ETH?

S474N, 2023-07-10:
Will the software update be via USB or OTA in the future?

Tarik2142, 2023-07-10:
If you have already flashed 1.0 via USB, then you can flash the following firmwares via OTA. USB flashing required only to update the partition table.

S474N, 2023-07-14:
Also have problem with upgrading Coordinator firmware:
[image]

[image]

S474N, 2023-07-14:
[sergge1](https://github.com/sergge1): step by step a couple of contributions higher, by short-circuiting two pins on the board. And yes, coordinator have access to internet.

Tarik2142, 2023-07-14:
This error means that the coordinator does not have access to the Internet.
Are you using a static address?

S474N, 2023-07-14:
Yes, static and trust me, have access to internet. How else would he find out that there is a new version for zigbee?

[image]

sergge1, 2023-07-14:
I guess Taras trust you as i do, however that is the code of the error, e.g. error with that code can't appear  for some other reason

S474N, 2023-07-14:
Is way, how can I check, that it have access to internet?

Or can I flash it with another way? (flasher link doesn't exists in doco).

S474N, 2023-07-14:
OK, tested DHCP and another vlan :D  
[image]

Now:
[image]
[image]

So thank you for cooperating!

sergge1, 2023-07-14:
> OK, tested DHCP and another vlan :D
> 
> Now: !
> 
> So thank you for cooperating!

So what was the problem? Is it initial vlan or is it DHCP/Static?
Just to trace and try to reproduce and fix it if it relates to DHCP.
Meanwhile, I've updated the link to the flashing tool in the [manual](https://smlight.tech/manual/slzb-06/guide/flashing-and-updating/updating-cc2652p.html#over-the-air-ethernet-update); thank you for pointing me out.

S474N, 2023-07-14:
Don't know, where is problem, because now have old VLAN and STATIC IP and it works:
[image]

sergge1, 2023-07-14:
> Don't know, where is problem, because now have old VLAN and STATIC IP and it works:
May be it was some glitch with the network - e.g. not stable internet access . Thanks for feedback

S474N, 2023-07-14:
I don't think it would be in the network, the connectivity is FTTH :D

[image]

[image]

Tarik2142, 2023-07-14:
I think that you previously had an error in the subnet mask, so the coordinator had problems accessing the Internet. In this firmware update, the check of this field was fixed

sergge1, 2023-07-29:
added information about documentation to the manual https://github.com/smlight-dev/slzb-06-manual/commit/26202dabab640dfd325345f761ea16f51554d3b8

closed 2023-07-29 by sergge1