smlight-dev/slzb-06-firmware #28: Error when trying to flash

neturmel, 2023-05-14:
I'd like to try the new 1.0.0beta firmware but I am not able to flash it. UART drivers are installed and I sucessfully flashed other esp32 devices.

I tried the webflasher and the windows esp-flasher-x86.exe (with admin rights) and three different USB cables, 
but I only get:

Unexpected error: ESP is not in flash boot mode. If your board has a flashing pin, try again while keeping it pressed.

or 

timeout

or 

permission denied on COMX port.

Could the device explicitly set to ESP-flash mode? Am I missing something?

Tarik2142, 2023-05-16:
Can you try on another computer?

Nico83500, 2023-05-16:
I flashed mine yesterday (ESP_Flasher, USB-C to USB-A 2 meters, Windows 11 x64 Pro + last UART drivers). Webflasher didn't work for me.

sergge1, 2023-05-20:
> I flashed mine yesterday (ESP_Flasher, USB-C to USB-A 2 meters, Windows 11 x64 Pro + last UART drivers). Webflasher didn't work for me.

Are there any possible reasons why web did not wotk for you? Just tested and works fine for me.

neturmel, 2023-05-21:
Was busy until now, just tried another computer - and it works with esp-flasher.

Nico83500, 2023-05-22:
I tried 2 computers (both on Windows 11 Pro), same issue. I also tried 3 different browsers (Chrome, Edge and Firefox).
Can I flash again with Web flasher if I already flashed it with ESP_Flasher ?
Thank you.

Nico83500, 2023-05-22:
@Tarik2142 I'll try again and I'll let you know but yes there was an error (reset device or press boot button while you click flash, I don't remember exactly)

Nico83500, 2023-05-25:
Ok so the error was "Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL"
I tried another computer, same issue with 2 different USB ports but the third worked. So flash seems to be OK. But ESP32 free heap has the same value than ESP_Flasher (287 KiB) and version 0.9.9 was 296 KiB. How can I be sure webflasher flashed my device (and I'm not using the ESP_Flasher version) ? Thank you !

Tarik2142, 2023-05-26:
287 KiB free heap is normal for firmware 1.0
Does your ESP_Flasher work with any USB and the web flasher only with one? Did I understand correctly?

Nico83500, 2023-05-26:
Yes you're right, ESP_Flasher works with any USB port but webflasher works only with 1 USB port

gabberpocky, 2023-06-11:
I'm struggling with the same thing. On both computers I have available I cannot do a USB ESP32 flash by any method. Both the web flasher and ESP-Flasher fail with the same error about a failure to initialise the bootloader. Holding down the button while booting makes no difference. I have tried multiple USB cables and every USB port in the computer, no difference at all.

After all these failed attempts I gave up and flashed 1.0 beta OTA, which worked but predictably Zigbee OTA updates don't work so I would like to do it properly. Any ideas?

Tarik2142, 2023-06-12:
@gabberpocky 
You can force your device into bootloader mode.
Remove the back cover and short the contacts marked in the photo, tweezers or a jumper wire do. After that, you can connect the device to USB and flash.
[image]

P.S Never give up

gabberpocky, 2023-06-12:
> @gabberpocky You can force your device into bootloader mode. Remove the back cover and short the contacts marked in the photo, tweezers or a jumper wire do. After that, you can connect the device to USB and flash. [image]
> 
> P.S Never give up

Thank you - fortunately not needed now (see my edit) but this will be helpful for anyone else having trouble.

closed 2023-06-19 by Tarik2142