smlight-dev/slzb-06-firmware #35: I used DIY hardware to program programming and he doesn't work. Did you limit its functionality, thanks

ahonxxx, 2023-06-18:
Title

Tarik2142, 2023-06-22:
Hi!
I don't understand what you can't do. I need more details.
Can you screenshot the error?

ahonxxx, 2023-06-22:
> 你好！ 我不明白你不能做什么。我需要更多细节。 能截图报错吗？

`[resetSettings] Start
[resetSettings] Led blinking done
[  5020][E][vfs_api.cpp:182] remove(): /config/configSerial.json does not exists or is directory
[  5022][E][vfs_api.cpp:182] remove(): /config/configSecurity.json does not exists or is directory
[  5028][E][vfs_api.cpp:182] remove(): /config/configGeneral.json does not exists or is directory
[  5036][E][vfs_api.cpp:182] remove(): /config/configEther.json does not exists or is directory
[  5045][E][vfs_api.cpp:182] remove(): /config/configWifi.json does not exists or is directory
[  5053][E][vfs_api.cpp:182] remove(): /config/system.json does not exists or is directory
[resetSettings] Config del done
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
Start
[hard reset] Entering hard reset mode
1
2
3
4
5
6
7
8
9
10`

Tarik2142, 2023-06-22:
Judging by the log, you have pressed the button

ahonxxx, 2023-06-22:
https://smlight.tech/manual/slzb-06/flasher/
After downloading the program through the above website, the serial port outputs this information
I checked all IO ports and there was no problem

Development board 
ESP32-DevKitC 
Externally connected  lan8720a

Tarik2142, 2023-06-22:
button should be pulled up
This repository contains the firmware for the SLZB-06 device, I understand you are trying to make DIY device?

ahonxxx, 2023-06-22:
Or the same, pull up that pin?

fighthebyte, 2023-07-02:
> pull up that pin?
> 
> I tried the bin file and it got this message
> 
> `rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT) invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 ets Jul 29 2019 12:21:46`

I had the problem,you need to pull up the BTN pin,in order to make it work.

ahonxxx, 2023-07-03:
> I had the problem,you need to pull up the BTN pin,in order to make it work.
I don't understand what you mean, can you elaborate, thank you

sergge1, 2023-07-03:
The issue tab is for issues with firmware designed for specific hardware.

ahonxxx, 2023-07-07:
> > 拔起那个引脚？
> > 我尝试了 bin 文件，它收到了这条消息
> > `rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT) invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 invalid header: 0x6d650065 ets Jul 29 2019 12:21:46`
> 
> 我遇到了问题，你需要拉起 BTN 引脚，才能使其工作。

> I had the problem,you need to pull up the BTN pin,in order to make it work.

OK, it's ready to work