smlight-dev/slzb-06-firmware #42: Zigbee OTA

fighthebyte, 2023-07-02:
Hi,
i build dev firmware [v1.0.0-b] for custom device and is working perfectly, except Zigbee OTA.
I find the script part of code where modalConstructor("flashZB"); was commented,and now firmware downloads,but no installation process, is anything i can do to have this feature?
Code for this feature was deleted from repo?

sergge1, 2023-07-03:
This feature is in the development stage for internal testing purposes and has not been released yet. Instead, use the final release version 0.9.9.

closed 2023-07-03 by sergge1

fighthebyte, 2023-07-03:
@sergge1 I have one SMLIGHT SLZB-06 working like main coordinator.
I have firmware 0.9.9 on it,happy about it.
If i will install v1.0.0-b, Zigbee OTA will work?

sergge1, 2023-07-03:
> @sergge1 I have one SMLIGHT SLZB-06 working like main coordinator. I have firmware 0.9.9 on it,happy about it. If i will install v1.0.0-b, Zigbee OTA will work?

Yes, it will work. 
But it needs to be flashed only via an online flasher (otherwise, it will not work - https://smlight.tech/manual/slzb-06/flasher/). 

and most important - it is beta for test purposes so I can't 100% guarantee it ALWAYS works correctly. So far, we are testing it.

fighthebyte, 2023-07-03:
@sergge1 thank you, waiting for release, to flash to my another devices and have Zigbee OTA working.

sergge1, 2023-07-03:
> @sergge1 thank you, waiting for release, to flash to my another devices and have Zigbee OTA working.

Why you cant use dev version for that?

fighthebyte, 2023-07-03:
> This feature is in the development stage for internal testing purposes and has not been released yet. Instead, use the final release version 0.9.9.

I flashed on my main device SMLIGHT SLZB-06, but i have some customs devices where i want to flash this firmware.I succed but no ZigBee OTA,this is the single feature i need curently.