smlight-dev/slzb-06-firmware #47: cannot join devices on new slzb-06

Cowh3adDK, 2023-08-04:
Hello, i already have one slzb-06 and 1 usb coordinator, i bought another slzb-06 to start another network, and did as it says in the guide, but the new coordinator will not join devices, i have been stuck on this issue for a few weeks now, i have tried reflashing firmware changing panids, zigbee channels, it will not work, tried installing zigbee2mqtt on another machine and nothing works. please help me, i dont know what more to try and fix this.  :)

sergge1, 2023-08-04:
can you please send us a message to the [email](mailto:info@smartlight.me) ?

> Hello, i already have one slzb-06 and 1 usb coordinator, i bought another slzb-06 to start another network, and did as it says in the guide, but the new coordinator will not join devices, i have [...]

Cowh3adDK, 2023-08-04:
> Hello, i already have one slzb-06 and 1 usb coordinator, i bought another slzb-06 to start another network, and did as it says in the guide, but the new coordinator will not join devices, i have [...]

done :)

sergge1, 2023-08-10:
Gentlemen, I am closing this issue here as this Issue tracker is dedicated to ESP32 firmware only. We moved with you to email correspondence instead.

closed 2023-08-10 by sergge1