smlight-dev/slzb-06-firmware #48: HA integration with Z2M + another Mosquitto together.

S474N, 2023-08-06:
I have my own Mosquitto server and as second device I have HA server with integration Z2M. 

If I set communication with HA, then device will offline on Mosquitto.

Is it bug or feature? :)

sergge1, 2023-08-10:
Please submit the ticket to the support system (see manual).
The issue is unclear - what do you mean under `HA server with Zigbee socket integration`? But from what I understood (if correctly)  - you can't run both Z2M and ZHA simultaneously with the coordinator.

sergge1, 2023-08-10:
[image]

S474N, 2023-08-14:
Hi, this integration in HA together with Z2M:
[image]

Tarik2142, 2023-08-16:
ZHA can work simultaneously with Z2M but with different coordinators, they do not work simultaneously with one coordinator. Zigbee networks should also be different so they don't interfere with each other

S474N, 2023-08-16:
Yes, I have only one piece ;)