smlight-dev/slzb-06-firmware #49: REQ error: http_code -1 during Zigbee OTA update

Worren, 2023-08-15:
Hi,
i have updated to esp firmware 1.0.1-dev . If I press the "Check for Updates" Button under "Zigbee OTA update", i get a list of three firmware options:
Coordinator Firmware Revision: 20230507
Coordinator Firmware Revision: 20221226
Router Firmware Revision: 20221102

If I try to update i get always an "REQ error: http_code -1" Error.

The device is connected in Lan mode (Wifi Disabled) and runs CC2652P firmware rev: 20221226

How can I further diagnose the problem?

Tarik2142, 2023-08-16:
Сoordinator does not have access to the Internet. This is required to download the firmware

mickeman, 2023-08-16:
I think i had the same problem when I had manually entered my ip/gateway settings.
If you have that try and set it to DHCP instead and retry the update.

Worren, 2023-08-16:
> I think i had the same problem when I had manually entered my ip/gateway settings. If you have that try and set it to DHCP instead and retry the update.

Thank you very much. This is it. If i enable DHCP i can update the firmware. I would never have thought of that. I would consider this as a bug.

Bonus Question: The firmware update window shows me a router firmware of 2022. Is this an old designation for the newer coordinator firmware?

Worren, 2023-08-16:
Gateway Address is correct.

sergge1, 2023-08-17:
@Worren can you please tell me the model and manufacturer of your router, to which the adapter is connected? Is your adapter connected over WiFi or Ethernet?

Worren, 2023-08-17:
Fritzbox7490 connected via Ethernet

aperezva, 2023-08-17:
Hi , I  have received the new dev ice, but I can´t see Zigbee OTA options on it. I´m in a 0.9.9  Esp32 version.

How can I know with zigbee coordination version has my devices?

sergge1, 2023-08-18:
@aperezva 
In order to get Zigbee updates options the device must be flased over USB once (all other updates will be available via web interface again). To do that, please follow these steps (4 min in total):
1. Connect your device to PC via USB-A - Type-C cable
2. Go to https://smlight.tech/manual/slzb-06/flasher/ in your browser
3. Select firmware version `1.0.1-dev` and press `Update firmware`. Follow a couple of questions and wait till notification about successful flashing.

After that, within your adapter's web interface, you will get `Zigbee OTA Update` - so you can flash your Zigbee SoC right from the web interface

aperezva, 2023-08-18:
Thanks solved.
Another question , I want to migrate from Son off Dongle coordinator to SMILGHT SLZB06.  Is there any manual or intructions for it?. May I need to repair the devices?

Thanks @sergge1

Tarik2142, 2023-08-22:
@aperezva  https://www.zigbee2mqtt.io/guide/faq/#how-do-i-migrate-from-one-adapter-to-another

aperezva, 2023-08-22:
Hi. 

I have already installed the coordinator. And it seems it s working. But I can see that green les is not blinking. We could only see it on a seconds. But I have only a yellow led on

any problem or configuration wrong?

thanks

landiuk, 2023-10-14:
@Worren Do you still have the firmware file for 1.0.1-dev ?

If you do, could you share the file as there are a number of people who are looking for it as the firmware link for 1.0.1-dev is dead. (see https://github.com/smlight-dev/slzb-06-firmware/issues/64)

mminehanNZ, 2023-10-14:
You could contact him using a burner email account (e.g. Temp Mail).

Worren, 2023-10-15:
Sorry, I have searched for it but no success.

Krazy998, 2023-11-11:
> If you have that try and set it to DHCP instead and retry the update.

The issue seems to be related to not setting DNS settings. When you disable DHCP and set static IP/mask the DNS settings are set to 0.0.0.0 so it cannot reach the internet. 

We need a ability to set DNS settings manually as well.