smlight-dev/slzb-06-firmware #50: reporting WiFi connection, although ethernet is used

Peter915671, 2023-08-16:
I noticed that the slzb-06 status screen suggests I am using a WiFi connection, although I connect via the IP-address that's configured for the ethernet connection.
Not sure what's causing this, but it is confusing.

[image]
[image]
[image]

Tarik2142, 2023-08-16:
The subnet mask is not set. Set the mask to 255.255.255.0

Peter915671, 2023-08-16:
Oh!
I thought that was the default, and I didn’t have to change that.
Changed.
Is all sorted now.
Thanks for the fast response.