smlight-dev/slzb-06-firmware #53: SLZB-06 latest Der (1.0.1) Firmware?

clipse2004, 2023-08-29:
Hello all, 

At the moment I use the SLZB-06 with the dev version 1.0.0.
On your site it says something about Dev Version 1.0.1 but where can I find it? When I click on the link, it only says "Page not found". 
Were there problems with this version? Will it be online again?

Merci

jesse-kaufman, 2023-09-03:
Still appears to be broken .. the online manual links to this repository talking about OTA updates, but there isn't a 1.0.x anywhere I can find. I'd love to be able to update my 2nd SLZB-06 with the firmware that supports OTA updates, as it's located out in my garage.

Arn0uDz, 2023-09-06:
I'm running the 1.0.1 dev firmware (sadly I don't have the firmware.bin anymore) and today I experienced my first Z2M disconnect. That lead me to here and maybe i'm going to try downgrading if it keeps happening.

error 2023-09-06 00:29:12: Adapter disconnected, stopping
info  2023-09-06 00:29:12: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2023-09-06 00:29:12: Disconnecting from MQTT server
info  2023-09-06 00:29:12: Stopping zigbee-herdsman...
error 2023-09-06 00:29:12: Failed to stop Zigbee2MQTT

Tarik2142, 2023-09-07:
> I'm running the 1.0.1 dev firmware (sadly I don't have the firmware.bin anymore) and today I experienced my first Z2M disconnect. That lead me to here and maybe i'm going to try downgrading if it [...]

It looks like a network problem

Tarik2142, 2023-09-07:
We are working on a new firmware version that will be available for download later. Stay tuned for updates in the repository

closed 2023-09-07 by Tarik2142