smlight-dev/slzb-06-firmware #54: impossible to appreat

jerome6994, 2023-09-03:
Good morning
I recently got the SLZB-06 POE module.
I used to have a Conbee II dongle.
I have almost 200 devices on my zigbee network with 80 routers and the rest are end devices.
I installed zigbee2mqtt in docker almost from the beginning.
I migrated from Conbee to SLZB-06 two days ago.
During the operation I lost modules in my network (aqara MCCGQ11LM door opening, aqara motion sensor
RTCGQ11LM, profalux roller shutter NSAV061) the rest of my past configuration seems operational.
I tried to re-apply the lost sensors or roller shutter but nothing can be done, no appliance works!
I even took new sensors with no additional success!

How do I pair lost items, especially my roller shutter?
How many devices possible with the POE module?

Thanking you for your valuable help

z2m version : 1.33

Jerome
France

pc1246, 2023-10-08:
Hi
Did you solve this meanwhile? I face the same/similar issue.
Regards Christoph

jerome6994, 2023-10-08:
Good morning

It seems that the key is made for 150 devices maximum!
to solve this you need 2 keys it seems but I have not yet understood how to set up this system without having 2 different networks which I do not want to do because my house poses a problem precisely in the mesh otherwise I would not have so many devices

pc1246, 2023-10-09:
Hmmmm
Thank you for your answer. I misunderstood maybe your post.
I thought you have the same problem like me. I can not pair any device, as the interview fails.
Thank you again and regards
Christoph

mminehanNZ, 2023-10-12:
Hi Jerome,
I am trying to also migrate from Conbee II to SLZB-06 but am having problems.
Did you have to re-pair every single device? Did you have to power down mains powered devices, and power them on and pair, one by one? Any advice?
Marty

jerome6994, 2023-10-13:
Good morning

So I did the pairing of 0, retrieving the configuration file did not work. Finally I have a configuration with the separate device and group file.
My biggest problem is having my 205 devices on the POE key! because limitation at 150 but I don't want to make 2 separate networks

smlight-dev-223942, 2023-12-10:
> Good morning
> 
> So I did the pairing of 0, retrieving the configuration file did not work. Finally I have a configuration with the separate device and group file. My biggest problem is having my 205 devices on the POE key! because limitation at 150 but I don't want to make 2 separate networks

have you solved your question somehow already?