smlight-dev/slzb-06-firmware #63: [Feature] Wireguard support

misuzu, 2023-09-25:
Would be awesome to be able to connect slzb-06 to HA in remote locations using this VPN protocol. 
ESPHome has this feature: https://github.com/esphome/esphome/pull/4256

misuzu, 2023-09-26:
> можете пояснити як воно має працювати?

Wireguard дозволяє створити зашифрований тунель між двома пристроями, у кожного пристрою є своя пара приватний/публічний ключ (як в ssh), та своя локальна IP-адреса. Тут є гарний опис як воно працює: https://www.wireguard.com/#conceptual-overview

Для підключення координатору до HA, на сервері з HA повинна бути публічна IP-адреса та сконфігурований Wireguard-тунель з публічною частиною ключа координатора.
На самому координаторі необхідно мати публічну частину ключа сервера, його публічну IP-адресу та UDP-порт тунелю Wireguard.
Після підключення координатора до сервера через Wireguard, HA має можливість підʼєднатися до координатора використовуючи локальну IP-адресу координатора.

> Звідки беруться ключі

Ключі можна згенерувати за допомогою `wg genkey`: https://www.wireguard.com/quickstart/#key-generation

> що потрібно вписувати в конфігурацію?

Приватний ключ пристрою, публічний ключ сервера, IP-адреса сервера, UDP-порт сервера, локальна IP-адреса тунелю з маскою.
Приклади для сервера та клієнта є тут: https://www.wireguard.com/#cryptokey-routing
Також в доці ESPHome про це трохи написано https://deploy-preview-2948--esphome.netlify.app/components/wireguard

> Вірно я розумію, що за допомогою цього VPN можна підключити віддалений координатор до HA через інтернет?

Так

Tarik2142, 2023-09-27:
Спробуйте цей білд [ota.zip](https://github.com/smlight-dev/slzb-06-firmware/files/12741548/ota.zip)

misuzu, 2023-09-27:
> Спробуйте цей білд [ota.zip](https://github.com/smlight-dev/slzb-06-firmware/files/12741548/ota.zip)

Це було швидко!

В логах щось є, але координатор не під'єднується до серверу Wireguard. Десь можна подивитись на зміни в коді? Можливо я щось підкажу.
```
setup|start
ConfigHelper|LittleFS mounted
ConfigHelper|load config
ConfigHelper|deserializeJson
ConfigHelper|Ok
[ZBCHK]|Wrong answer
[ZBVER]|Unknown
Network|init
Network|ETH start
Network|EVENT_ETH_START
Network|EVENT_ETH_CONNECTED
Web|Webserver started
setup|done
Network|[MDNS] Started
Network|EVENT_ETH_GOT_IP|ETH MAC: A8:42:E3:46:A9:33 IPv4: 192.168.0.118 linkSpeed: 100Mbps
Network|[WG] Starting Wire Guard
Network|[WG] Wire Guard started
Network|[ZB] Starting server
TX|00 fe 06 41 80 00 02 01 02 07 01 c0 
Network|[ZB] Task heap: 4744
```
[image]

```
% sudo wg
interface: wg0
  public key: LbyU1D1GQsmoZByzRSsNfPp8b4uFzBl+A8R2Zx0kYkk=
  private key: (hidden)
  listening port: 51820

peer: VP4HHDN7SHw65+SFonbCdUq+MTuZbkxofjmH1uzfxik=
  allowed ips: 10.100.0.2/32
```

Tarik2142, 2023-09-28:
> Це було швидко!
>
> В логах щось є, але координатор не під'єднується до серверу Wireguard. Десь можна подивитись на зміни в коді? Можливо я щось підкажу.
>
> ```
> setup|start
> ConfigHelper|LittleFS mounted [...]

Я використав лібу https://github.com/ciniml/WireGuard-ESP32-Arduino але схоже, що вона не працює

Tarik2142, 2023-09-28:
Ви взагалі використовували цей компонент в esphome?

Tarik2142, 2023-09-28:
@misuzu спробуйте цей білд [ota.zip](https://github.com/smlight-dev/slzb-06-firmware/files/12750408/ota.zip)

misuzu, 2023-09-28:
> Ви взагалі використовували цей компонент в esphome?

Вони цю фічу тільки зарелізили, все працює, пакети ходять.
Тестував з таким конфігом:

```yaml
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

wireguard:
  address: 10.100.0.2
  private_key: KAFIRAPVvnGZUko+0bw0tPOobseJUaC/eJkq3H3tImI=
  peer_endpoint: 192.168.0.101
  peer_public_key: LbyU1D1GQsmoZByzRSsNfPp8b4uFzBl+A8R2Zx0kYkk=

  # Optional netmask (this is the default if omitted)
  netmask: 255.255.255.255

  # Optional endpoint port (WireGuard default if omitted)
  peer_port: 51820

  # Optional list of ip/mask (any host is allowed if omitted)
  peer_allowed_ips:
    - 0.0.0.0/0

  # Optional keepalive (disabled by default)
  peer_persistent_keepalive: 25s

time:
  - platform: sntp
    id: sntp_time
```

```
[15:54:18][I][wifi:560]: WiFi Connected!
[15:54:18][C][wifi:382]:   Local MAC: D4:D4:DA:84:73:08
[15:54:18][C][wifi:383]:   SSID: 'Giovanna'
[15:54:18][C][wifi:384]:   IP Address: 192.168.0.112
[15:54:18][C][wifi:386]:   BSSID: 34:60:F9:6C:2B:C5
[15:54:18][C][wifi:387]:   Hostname: 'esp32-bluetooth-proxy-847308'
[15:54:18][C][wifi:389]:   Signal strength: -66 dB ▂▄▆█
[15:54:18][C][wifi:393]:   Channel: 5
[15:54:18][C][wifi:394]:   Subnet: 255.255.255.0
[15:54:18][C][wifi:395]:   Gateway: 192.168.0.1
[15:54:18][C][wifi:396]:   DNS1: 192.168.0.1
[15:54:18][C][wifi:397]:   DNS2: 192.168.0.1
[15:54:18][D][wifi:569]: Disabling AP...
[15:54:18][C][sntp:028]: Setting up SNTP...
[15:54:18][D][wireguard:037]: initializing WireGuard...
[15:54:18][I][wireguard:053]: WireGuard initialized
[15:54:18][C][ota:097]: Over-The-Air Updates:
[15:54:18][C][ota:098]:   Address: esp32-bluetooth-proxy.local:3232
[15:54:18][W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[15:54:18][C][api:025]: Setting up Home Assistant API server...
[15:54:18][I][app:062]: setup() finished successfully!
[15:54:18][D][wireguard:099]: WireGuard remote peer is offline (latest handshake timestamp not available)
[15:54:18][D][wireguard:203]: WireGuard is waiting for system time to be synchronized
[15:54:18][D][wireguard:203]: WireGuard is waiting for system time to be synchronized
[15:54:18][I][app:102]: ESPHome version 2023.9.1 compiled on Sep 28 2023, 15:51:52
[15:54:18][I][app:104]: Project esphome.bluetooth-proxy version 1.0
[15:54:18][C][wifi:546]: WiFi:
[15:54:18][C][wifi:382]:   Local MAC: D4:D4:DA:84:73:08
[15:54:18][C][wifi:383]:   SSID: 'Giovanna'
[15:54:18][C][wifi:384]:   IP Address: 192.168.0.112
[15:54:18][C][wifi:386]:   BSSID: 34:60:F9:6C:2B:C5
[15:54:18][C][wifi:387]:   Hostname: 'esp32-bluetooth-proxy-847308'
[... 63 lines cut ...]
[15:54:20][I][wireguard:226]: WireGuard connection started
[15:54:20][D][wireguard:232]: configuring WireGuard allowed IPs list...
[15:54:20][D][esp-idf:000]: I (10398) esp_wireguard: add allowed_ip: 0.0.0.0/0.0.0.0
[15:54:20]
[15:54:20][D][wireguard:240]: allowed IPs list configured correctly
[15:54:20][W][component:204]: Component sntp.time took a long time for an operation (0.10 s).
[15:54:20][W][component:205]: Components should block for at most 20-30ms.
[15:54:23][I][wireguard:089]: WireGuard remote peer is online (latest handshake 2023-09-28 15:54:20 UTC)
[15:54:33][D][wireguard:092]: WireGuard remote peer is online (latest handshake 2023-09-28 15:54:20 UTC)
[15:54:43][D][wireguard:092]: WireGuard remote peer is online (latest handshake 2023-09-28 15:54:20 UTC)
```

```shellsession
% sudo wg
interface: wg0
  public key: LbyU1D1GQsmoZByzRSsNfPp8b4uFzBl+A8R2Zx0kYkk=
  private key: (hidden)
  listening port: 51820

peer: VP4HHDN7SHw65+SFonbCdUq+MTuZbkxofjmH1uzfxik=
  endpoint: 192.168.0.112:51046
  allowed ips: 10.100.0.2/32
  latest handshake: 2 minutes, 18 seconds ago
  transfer: 692 B received, 508 B sent
```

misuzu, 2023-09-28:
> @misuzu спробуйте цей білд [ota.zip](https://github.com/smlight-dev/slzb-06-firmware/files/12750408/ota.zip)

Конфіг той самий, результат теж - з'єднання нема:

```
setup|start
ConfigHelper|LittleFS mounted
ConfigHelper|load config
ConfigHelper|deserializeJson
ConfigHelper|Ok
[ZBCHK]|Connection OK
Network|init
Network|ETH start
Network|EVENT_ETH_START
Network|EVENT_ETH_CONNECTED
Web|Webserver started
setup|done
Network|[MDNS] Started
Network|EVENT_ETH_GOT_IP|ETH MAC: A8:42:E3:46:A9:33 IPv4: 192.168.0.117 linkSpeed: 100Mbps
Network|[WG] Starting Wire Guard
Network|[WG] WireGuard init ok
Network|time: Friday, September 29 2023 01:05:4
Network|[WG] WireGuard connection started
Network|EVENT: 0
Network|[ZB] Starting server
Network|EVENT: 3
Network|[ZB] Task heap: 4440
```

```
% sudo wg
interface: wg0
  public key: LbyU1D1GQsmoZByzRSsNfPp8b4uFzBl+A8R2Zx0kYkk=
  private key: (hidden)
  listening port: 51820

peer: VP4HHDN7SHw65+SFonbCdUq+MTuZbkxofjmH1uzfxik=
  allowed ips: 10.100.0.2/32
```

ESP32 firmware ver: 1.1.2-dev (Sep 28 2023)

Tarik2142, 2023-10-03:
Доброго дня, я мав успіх із цим білдом [ota.zip](https://github.com/smlight-dev/slzb-06-firmware/files/12793479/ota.zip)
[image]
Тут захардкожено деякі параметри, а саме:
netmask 255.255.255.255
peer_allowed_ips 0.0.0.0/0
це ок, чи вони також потрібні в меню налаштувань?

misuzu, 2023-10-03:
> Доброго дня, я мав успіх із цим білдом [ota.zip](https://github.com/smlight-dev/slzb-06-firmware/files/12793479/ota.zip)

Доброго, я спробую пізніше, дякую!

> це ок, чи вони також потрібні в меню налаштувань?

Я думаю ок, принаймні я не бачу сенсу в інших значеннях

misuzu, 2023-10-03:
> спробую пізніше

Наче працює як треба, zigbe2mqtt через тунель підключається, пристрої бачить, з NAT теж проблем нема.
Єдине що після ввімкнення режиму WiFi координатор перестає працювати, допомагає лише перепрошивка через USB

Tarik2142, 2023-10-04:
> > спробую пізніше
> 
> Наче працює як треба, zigbe2mqtt через тунель підключається, пристрої бачить, з NAT теж проблем нема. Єдине що після ввімкнення режиму WiFi координатор перестає працювати, допомагає лише перепрошивка через USB

це ок, в цьому білді був поламаний wifi.
Тоді я закриваю цей запит, очікуйте на новий реліз, він буде включати цей функціонал

closed 2023-10-04 by Tarik2142