smlight-dev/slzb-06-firmware #64: Dev firmware link dead

landiuk, 2023-09-30:
Have dev ESP firmware been removed?

Latest dev release link in readme.md returns 404 error

mminehanNZ, 2023-10-06:
Yes....I have the exact same issue.
Any idea when the firmware will be available?

renedis, 2023-10-07:
Same for me. Seems like the firmware is not uploaded.

MaMerz, 2023-10-09:
I really also would like to use the 1.0.0 or 1.0.1 ...
@sergge1 Any Idea where its gone?

landiuk, 2023-10-14:
I have asked Worren in the above thread if he still has the file and if he does, could he share it.

bikeymouse, 2023-10-14:
@smlight-dev-223942 Any chance there will be a new dev or release version realeased soon? 
It has been quite a while now.
Thanks for your support!

Luligu, 2023-10-14:
> I have asked Worren in the above thread if he still has the file and if he does, could he share it.

Thanks would be great to at least update the coordinator firmware.

Luligu, 2023-10-14:
If anyone have a dev from 1.0 to enable the auto update of firmware

GeFrisbi, 2023-10-14:
Hi friends, ping me at geofrisbi at gmail.com

landiuk, 2023-10-14:
> Hi friends, ping me at "geofrisbi at gmail.com" - i will try to help you.

Forgive me for being cautious, but you only created your account on github 5 minutes ago

GeFrisbi, 2023-10-14:
So? :) It's up to you; I just wanted to help. Then, better write to their ticketing system..

mminehanNZ, 2023-10-14:
You could contact him using a burner email account (.e.g Temp Mail).

landiuk, 2023-10-17:
> I have asked Worren in the above thread if he still has the file and if he does, could he share it.

Worren has replied, he doesn't have the file

smlight-dev-223942, 2023-10-17:
Hi gentlemen, sorry for the disappointment. 
We are still working on the update; there were issues with the beta development version and that is why it has not been released yet. Meanwhile, as advised by GeFrisbi above, please send support requests (you have the links so the engineers can help you one by one. you have a link to the ticketing system with your device).

Arn0uDz, 2023-11-10:
When you go to this link: https://smlight.tech/flasher/

There is an option to flash "2.0.1-dev". Anyone tried this?

Krazy998, 2023-11-11:
i tried the 2.0.3-dev and it works. There are issues with a couple of things broken. But overall fine.

Seems couple of extra features like wireguard vpn support. I tried the zigbee flashing capability and it throws a error. Heres a screenshot of the dev build for anyone curious.

[image]

Arn0uDz, 2023-11-11:
Thanks for the info. 

It's sad that zigbee flashing is advertised for this product as a feature but it doesn't even work. The software development is also lacking and you get no info on what's going on. Wish I didn't spend this much money on the product.

GeFrisbi, 2023-11-11:
> Thanks for the info.

Did you flash with the link you shared before? I used it and Zigbee update works for me.  Just updated smoothly

Krazy998, 2023-11-11:
> It's sad that zigbee flashing is advertised for this product as a feature but it doesn't even work

This is a dev build which isn't actually publicly released and it does actually work. The current stable version allows you to flash the zigbee module remotely by using the following guide: https://smlight.tech/manual/slzb-06/guide/flashing-and-updating/updating-cc2652p.html#manual-over-the-air-ethernet-update

The new build seems to offer a feature to automatically check regularly and update during a certain window. A neat feature.

Arn0uDz, 2023-11-11:
I tried the zigbee flash when I was on dev build 1.0.1 and all the different options and guides didn't work for me. It's a couple months ago now but I think it gave a file mismatch error or something. 

Maybe it was a driver issue since I'm using an AMD chipset.

Tarik2142, 2023-11-13:
> i tried the 2.0.3-dev and it works. There are issues with a couple of things broken. No issues with connectivity from zigbee2mqtt. Im happy to test bleeding edge, so will monitor the repo
> 
> EDIT: I noticed that the zigbee flashing doesnt work if you have static IP/mask/gateway set. It seems DNS settings are not possible to set, so it doesnt know how to reach the internet. We need a ability to set DNS settings statically.
> 
> Seems couple of extra features like wireguard vpn support. Heres a screenshot of the dev build for anyone curious.
> 
> [image]

Try the latest 2.0.5 build, it has google DNS 8.8.8.8, that should help

smlight-dev-223942, 2023-12-10:
closed as resolved

closed 2023-12-10 by smlight-dev-223942