smlight-dev/slzb-06-firmware #65: Router random disconnection for ZBMINIL2

amaduain, 2023-10-08:
Hi,
I'm using the SLZB-06 as a router (changed firmware to router) and I notices that a couple of SONOFF ZBMINIL2 go disconnected from time to time.
After some troubleshooting, I discovered that restarting the SLZB-06 the missing switches go back to live.
SLZB-06 Firmware : 0.9.9
Used with Z2MQTT and Home Assistant
Coordinator SONOFF ZB Dongle-P
Both switches are 4-5 meters away from the SLZB-06, I have other devices and same ZBMINIL2 connected to other routers and works flawleslly (mostly ikea tradfri)

Any other info, let me know.

Slava Ukraini!

Luligu, 2023-10-12:
HI, sorry to disturb. Can you telle me how you managed to enter boot loader mode? I have Firmware : 0.9.9 and I tried everything but never been able to make it. Waiting for the ESP32 release with web updater. Maybe you can explain me how to do it. My best

amaduain, 2023-10-12:
Try following the official guide:
https://smlight.tech/manual/slzb-06/guide/flashing-and-updating/updating-cc2652p.html#zigbee-auto-update-over-the-web-interface-recommended

In case of using usb you need to configure the usb mode and enable flashing.
You have the detailed instructions on the link

denveronly, 2023-10-13:
> Hi, I'm using the SLZB-06 as a router (changed firmware to router) and I notices that a couple of SONOFF ZBMINIL2 go disconnected from time to time. After some troubleshooting, I discovered that [...]

I guess it is a problem of Ezsp chip of sm and zbmini firmware

had similar on sonoff dongle-e which uses same chip as SM. Continuos disconnects of zbmini and aqara relays. Everything else was pretty fine

I got here to see if SM could do anything with disconnects with ezsp chips, but see that problem is still there sadly

Slava Ukraine!