smlight-dev/slzb-06-firmware #66: Socket Random failure and Zigbee Flash not possible

Coldness00, 2023-10-19:
Hello Dears,

Since a couple of days I have issue with the equipment as the Socket Disconnect randomly and sometime, refuse to be accessed through the webpage. Only solution is to unplug then plug back. (This is rare case)

I loose connexion of coordinator for a couple of sec then it comes back (time also for z2m to reboot)
I would say it happen 2 to 5 time a day.
Ver 0.9.9

I use this coordinator via LAN, running, Home Assistant, Mosquitto and Z2M as Docker

I though this might come from the old Zigbee Firmware I still use (CC2652PP1FRGZR), so I followed procedure (https://smlight.tech/manual/slzb-06/guide/flashing-and-updating/updating-cc2652p.html#over-the-usb-update) and when I try to flash either Flash Programm crach either it tells me: Unknown record type: 3

Could you please help me with the matter?
Thank you in advance.

Koumi460, 2023-10-25:
Hi,

Recently I am also experiencing similar issue. The socket randomly disconnects from z2m. But the web UI keeps working okay. I always go in to the web UI and restart Zigbee and ESP32 and then boot z2m back up and the problem is solved. But it seems to be happening a few times a week now and I don't know how to troubleshoot it. 
Also firmware 0.9.9 

Any help would be appreciated!

Coldness00, 2023-10-25:
Hello,

I was experiencing issue with one device, which was acting strange on command (laggy, unresponsive). I removed it from the network and since then, it has been running fine for more than 3days. 

Device: Aqara Valve thermostat E1 

Let see if this get confirmed but since all good.

Koumi460, 2023-10-26:
I do also have some Aquara devices, mostly wall switches. Maybe it is related. But the strange thing for me is that this started more recently and my zigbee network has been the same for over a year. Maybe something in Z2M updates? Is there a way to get logs out of slzb-06?

Coldness00, 2023-10-26:
I have other Aqara device surch as PIR and Magic Cube, and I don't have any issue with.

I would like to know also if we can pull logs out of this smart thing.

JC38, 2023-11-05:
For those who have the ransom socket failures, could you specify how the SLZB06 is powered ?

Koumi460, 2023-11-05:
> For those who have the ransom socket failures, could you specify how the SLZB06 is powered ?

Mine is via PoE switch.

Coldness00, 2023-11-05:
I use the usb port from my UPS Eaton 3S 700

JC38, 2023-11-06:
FYI, when I received mine, I wanted to use it with a USB power and in wireless. Unfortunately, I got plenty of network issues (socket failures) btw the device and z2mqtt. But everytime, the system was up, only a problem with the socket. Then, I plugged it in ethernet mode with Poe and no mor problem anymore. 
I can't say if the problem was on the USB power side or into the SLZB, but using ethernet Poe solved my problem. So at least for testing, I suggest everyone with a socket issue to test it behind a Poe port. 
@Koumi460 : did you try another Poe port or behind another Poe injector ?

Arn0uDz, 2023-11-06:
I'm using PoE since day 1 and have been getting the socket disconnect, so that's not the solution.

Koumi460, 2023-11-06:
> @Koumi460 : did you try another Poe port or behind another Poe injector ?

Yes, I have tried different PoE ports and no luck. Other devices on the PoE switch work flawlessly. I went the PoE route from the beginning because I read how many users have interference issues when using Wifi connected Zigbee coordinators. Also worth noting that I've now found out that the issue gets temporarilly fixed by restarting the Zigbee module on my SLZB-06 without the need to restart the ESP32.

Tarik2142, 2023-11-09:
For those who need to restart zigbee: your zigbee chip is crashing. This can occur for various reasons: too many interference in the network, too many devices, a bug in the firmware.
There are also known cases when some devices spam packets in the Zigbee network and overload the coordinator

Tarik2142, 2023-11-09:
> Hello Dears,
>
> Since a couple of days I have issue with the equipment as the Socket Disconnect randomly and sometime, refuse to be accessed through the webpage. Only solution is to unplug then plug [...]

Update ESP32 firmware https://smlight.tech/flasher/ and try zigbee OTA after that

Dabbadoeber, 2023-11-09:
Update ESP32 firmware https://smlight.tech/flasher/ and try zigbee OTA after that

Well would be nice if it worked.
i tried 2 SLZB-06 sticks none of them where able to connect
They are detected.
They are connected wtih usb-A to usb-C (tried different cables)
One stick was running fine until a couple of days ago when strange things happen in the Zigbee network, and connection from Z2M was gone but web interface was still  accessible
and sais that socket was connected, but Z2M would not boot because of no connection

Tarik2142, 2023-11-09:
> Update ESP32 firmware https://smlight.tech/flasher/ and try zigbee OTA after that
> 
> Well would be nice if it worked.
> 
> get the message: Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.
> 
> i tried 2 SLZB-06 sticks none of them where able to connect They are detected. They are connected wtih usb-A to usb-C (tried different cables)
> 
> Led sequence Blue blinking 2x Orange is on Green off Then Blue off Orange off Green on Blue blinking 3x Orange is on Green off Then All off

This means that the flasher cannot put the device into flash mode.
This happens on some PCs due to a bug in the driver of the USB-TTL chip manufacturer, at the moment there are 2 working solutions:
1. Try another PC
2. Put the device in manual firmware mode, but for this you need to remove the bottom cover

Dabbadoeber, 2023-11-09:
Yes busy on my macbook (in vmware windows) seems to be runing now

VeganBiker, 2023-12-10:
> This means that the flasher cannot put the device into flash mode. This happens on some PCs due to a bug in the driver of the USB-TTL chip manufacturer, at the moment there are 2 working [...]

How do you put it in manual firmware mode? I have no problem removing the bottom cover. I don't see the details in documentation

Coldness00, 2023-12-10:
Thank you for the feedback.
i used the webflasher, personnaly, to go to new dev version 2.0.8 and I updated Zigbee also.
Since then, it's far better. I do not have any loss of connection and network is more reliable, specially on my low power PIRs.

Koumi460, 2023-12-11:
> For those who need to restart zigbee: your zigbee chip is crashing. This can occur for various reasons: too many interference in the network, too many devices, a bug in the firmware. There are also known cases when some devices spam packets in the Zigbee network and overload the coordinator

Then it seems my zigbee chip is crashing. I have 54 devices in my zigbee network. When checking debug logs in Z2M there is about 2-3 messages per second average without any obvious spamming or issues. But I don't know how to check for packets spamming. How can I throubleshoot this further to find out what is crashing it?