smlight-dev/slzb-06-firmware #69: Firmware for SLZB-06M

RaveGun, 2023-11-11:
The link to the FW is not available.
https://github.com/smlight-dev/slzb-06m-firmware/releases/latest

Is the repository missing or private?

Thank you.

quenthal, 2023-11-11:
Hope this device is not abandoned?

Krazy998, 2023-11-11:
Its not abandoned. There has been a new dev build published to another repo.

Stable build can be found here: https://github.com/smlight-dev/slzb-06-firmware/releases/tag/v0.9.9

If you want the latest dev build (2.0.3 released yesterday) then you can flash via this: https://smlight.tech/flasher/

RaveGun, 2023-11-12:
Thanks!

This means that the same binary it is used for the ESP32 no matter what Zigbee chip is used?

Greetings.
.

RaveGun, 2023-11-12:
Oh, I found it now:
It was here:
https://github.com/smlight-dev/smlight-uniflasher-firmwares

Excellent, many thanks.

pc1246, 2023-11-17:
@RaveGun 
You are my hero!
Since the webflasher does not work for me,
I need to try the dev SW as my device does not pair with any device with 0.9.9.
Regards Christoph

pc1246, 2023-11-17:
Hmmm
But I am not able to flash these via the webpage.
Any hints?
Regards Christoph

Tarik2142, 2023-11-17:
> Hmmm But I am not able to flash these via the webpage. Any hints? Regards Christoph

What error are you getting?

pc1246, 2023-11-18:
Hi Tarik
Sorry, I did not see your message before.
If I am using any of the downloaded FW versions the flash process goes very quick, and afterwards the device comes back with 0.9.9.
When I try to use the web tools the device seems to restart multiple times and it finishes with:
"Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL."
With pressed button the same!
Regards Christoph

pc1246, 2023-11-18:
Hi
I did meanwhile some more investigation with ESPTOOL:
- chip_id could not be read, due to wrong boot mode (0x12)
- checking the images with image_info gave: A" fatal error occurred: This is not a valid image (invalid magic number: 0xff)"
I could try to use a linux based machine, instead of W11, but I do not see a chance!
Regards Christoph

Tarik2142, 2023-11-18:
> Hi Tarik Sorry, I did not see your message before. If I am using any of the downloaded FW versions the flash process goes very quick, and afterwards the device comes back with 0.9.9. When I try to [...]

this is a known problem, a bug in the usb-ttl chip windows driver, due to which the device does not enter boot mode on some PCs. Try webflasher on another PC

webknjaz, 2023-12-10:
@Tarik2142 are there any plans to open-source the firmware for 06M the same way is it is for 06? I'm exploring the coordinators and this would be an important data point for deciding which one to purchase.

Is the 06M version considered experimental (both software and hardware-wise)? https://www.zigbee2mqtt.io/guide/adapters/#experimental still lists it a such.

BoBeRzE, 2023-12-11:
@Tarik2142 I would also like to buy the SLZB-06M. There is advertising for open source firmware but the current firmware is not publicly available. For this reason I will not buy the device. That is not very trustworthy

Tarik2142, 2023-12-11:
> @Tarik2142 I would also like to buy the SLZB-06M. There is advertising for open source firmware but the current firmware is not publicly available. For this reason I will not buy the device. That is not very trustworthy

Everyone makes their own choice ¯\_(ツ)_/¯

BoBeRzE, 2023-12-11:
that's fine, but then please don't advertise with open source firmware if the current developments are not made publicly available

Tarik2142, 2023-12-11:
> that's fine, but then please don't advertise with open source firmware if the current developments are not made publicly available

Sounds fair to me

BoBeRzE, 2023-12-12:
oh how nice, censorship is practiced here