smlight-dev/slzb-06-firmware #70: FW 2.0.3-dev 111123 - web server authentication

landiuk, 2023-11-12:
Wasn't sure where to put this with the firmware being under the SMlight updater.

After flashing to FW 2.0.3-dev 111123 from 0.99 the web server authentication max password length has been reduced to 8 characters.

Setting password will accept more than 8 characters without error, but only the first 8 characters are retained.

Tarik2142, 2023-11-12:
Login and password fields do not accept more than 8 characters

landiuk, 2023-11-16:
Sorry for delay in responding wanted to check v0.99

v0.99 accepts 24 character password on web server authentication

Tarik2142, 2023-11-16:
> Sorry for delay in responding wanted to check v0.99
> 
> v0.99 accepts 24 character password on web server authentication

Yes I know. The question is, do you need 24 characters?
In 2.0, the maximum length of password and login is 8 characters

DerDreschner, 2023-11-29:
@landiuk : I guess it's a space limitation in flash memory. The whole wireguard stuff and other new features might use quite a bit of storage. Keep in mind that there are only 16Mb for all that great stuff.

@Tarik2142 : Could it be that there is a problem with special characters, too? I had some issues setting a randomized password from KeepassXC with logograms enabled ( # $ % & @ ^ ` ~ ). But as I switched browsers in between from firefox to chrome, I'm not sure if it's a firefox thing or maybe something on my end.

Tarik2142, 2023-11-29:
> @landiuk : I guess it's a space limitation in flash memory. The whole wireguard stuff and other new features might use quite a bit of storage. Keep in mind that there are only 16Mb for all that [...]

In the next release, I will increase the password length to 24 characters.
There are no restrictions on special characters, they should work

smlight-dev-223942, 2023-12-10:
> Personally I don't see why you would lower security, but if it's intentional then by all means close as not planned.

in release 2.0.11 (available in online flasher) - this has been solved, max number of characters increased to 24

closed 2023-12-10 by smlight-dev-223942