smlight-dev/slzb-06-firmware #72: SLZB-06 - 2.0.6 DNS Settings

Krazy998, 2023-11-14:
Hello!

Thanks for the updates to the firmware.

I understand with the new version you are hard coding 8.8.8.8 as a DNS server to make sure it is able to connect when static IP is set. However could you consider the following:

If set via DHCP - then set the DNS server that DHCP offers.
If set statically - offer IP / Mask / Gateway + 1 or 2 DNS server options. This will allow people to use internal DNS servers. 

The reason why forcing 8.8.8.8 doesn't really work for me is because I run a internal DNS server and offer internal NTP server as well.

Hopefully its a easy option to add ability to set DNS server as well

jonnycastaway, 2023-11-14:
Hi.

Same here! All NTP or DNS queries to external addresses are blocked. Only internal servers are used. So imho the ability to set own DNS while using static IP settings is a requirement.

closed 2023-12-07 by Tarik2142

BlastyCZ, 2023-12-07:
DNS has nothing to do with default gateway setting. If DNS server is in the same broadcast domain you don't even need to setup default gateway (or any routes). For example I need to solve DNS queries with my local DNS server because I want to use my local NTP servers. Internet access is not allowed in my IOT vlan.

Tarik2142, 2023-12-07:
> DNS has nothing to do with default gateway setting. If DNS server is in the same broadcast domain you don't even need to setup default gateway (or any routes). For example I need to solve DNS [...]

Device receives DNS from the router, a gateway is used for this

BlastyCZ, 2023-12-07:
No, if static address is set, router is not required and you have to set ip address of DNS server manually (default gateway too if needed). And btw default gateway has nothing to do with DHCP server. They are just in SOHO setups usually run on the same device.

BlastyCZ, 2023-12-08:
> Ok, I can do a test build with the ability to set DNS but I don't think that will help. Initially, manuall DNS setting was supposed to solve the problem of Internet access for those who use manual IP (and probably set the wrong gateway), but it does not work. Why access to the Internet? The device downloads the Zigbee firmware from our server during the update, and in order for the request to pass, need to get an IP from the site's domain.

Don't get my wrong I completely understand your point with internet access. But manual setting of DNS server is a basic setting which should be available at all circumstances even when dhcp client is on.

Tarik2142, 2023-12-08:
> > Ok, I can do a test build with the ability to set DNS but I don't think that will help.
> 
> Yes it will help in scenarios where users set static IP on the SLZB device. Happy to test when you have a build ready!

I will add DNS in dev-fw 2.0.10 so please wait a bit

Krazy998, 2023-12-12:
I can confirm that setting static DNS settings work fine in new 2.0.12 version and the device is able to query my local DNS server. 

Thank you!