smlight-dev/slzb-06-firmware #73: SLZB-06 - 2.0.6 SoC Temperature

Krazy998, 2023-11-14:
After updating to 2.0.x builds I noticed that the temp of the SoC seems to be extremely high - most likely in Fahrenheit instead of Celsius.

Maybe offer some sort of localisation option to switch between the two for our imperial colleagues :-)

[image]

jonnycastaway, 2023-11-14:
And also confirm, same here.

[image]

Tarik2142, 2023-11-14:
> After updating to 2.0.x builds I noticed that the temp of the SoC seems to be extremely high - most likely in Fahrenheit instead of Celsius.
> 
> Maybe offer some sort of localisation option to switch between the two for our imperial colleagues :-)
> 
> [image]

It's Celsius. Usually this high temperature when WIFI is on

jonnycastaway, 2023-11-14:
Ok, back to 0.9.9 the temperature is 24°C.
@Krazy998 i think you're right, the numbers are °F

Krazy998, 2023-11-14:
> Usually this high temperature when WIFI is on

No Wifi on here just PoE. When I had 0.9.9 the CPU temp was around 25-30c. So that is why I suspected its in °F. As @jonnycastaway mentioned - you can also see the high temp on the zigbee SoC as well.

jonnycastaway, 2023-11-15:
It also noticed the fact that, with 2.0.6-dev all my zigbee devices reacts much slower. multiple seconds delay. with 0.9.9 all is smooth and nearly instantly.

I didn't touched the Zigbee SoC  firmware.

Back on 0.9.9 all is working smooth again.

So it's also possible that 2.0.6-dev really generates a heavy load on the CPUs and the impact isn't only the temperature but also the speed of communication (maybe between ESP32 and Zigbee SoC or ESP32 to Ethernet). Maybe the CPU runs in a limitation because of the temperature to save the CPU.

Krazy998, 2023-11-15:
> also noticed the fact that, with 2.0.6-dev all my zigbee devices reacts much slower.

Ok Im really glad you said this. I am not the only one. I was trying to debug this thinking it was something I did. I am also seeing a 1-2 second zigbee delay with 2.0.x builds. 0.9.9 perfect.

That is a big deal for me, so I will move back to 0.9.9 until that issue is resolved.

jonnycastaway, 2023-11-15:
> > also noticed the fact that, with 2.0.6-dev all my zigbee devices reacts much slower.

> That is a big deal for me, so I will move back to 0.9.9 until that issue is resolved.

For me also because it makes no sense to have an PIR Sensor in the hallway that reacts when i had walked half through it. And i have many PIR Sensors (7)

Tarik2142, 2023-11-15:
@jonnycastaway @Krazy998
Thanks for reporting this. I want to have a quick communication with you for further tests and solving these problems.
Could you email us at support@smartlight.me ?

jonnycastaway, 2023-11-15:
> @jonnycastaway @Krazy998 Thanks for reporting this. I want to have a quick communication with you for further tests and solving these problems. Could you email us at [support@smartlight.me](mailto:support@smartlight.me) ?

done, you have mail

Krazy998, 2023-11-15:
> Could you email us at [support@smartlight.me](mailto:support@smartlight.me) ?

likewise - happy to test further

xvisor16, 2023-11-16:
Interesting, here the temp seems quite ok actually while it's a little higher then before it's not much:
[image]
PoE powered, WiFi off. No USB connection.

I did however needed to reapply the 20230507 zigbee coordinator firmware to get the version visible.

landiuk, 2023-11-16:
I notice that every time you update the ESP it loses the zigbee version and last check is always 29th Sept

Tarik2142, 2023-11-16:
> I notice that every time you update the ESP it loses the zigbee version and last check is always 29th Sept

The problem with wiping ZB version will be solved as soon as we start publishing the OTA files, with the OTA update of the esp32, flashed ZB version will be saved.
Regarding 29th Sept, now this is a static text because the function is under development

gtalusan, 2023-11-17:
Hello, I'm running 2.0.6-dev on my 06m and I noticed that temperatures were hotter than normal.  I'm on PoE - on 0.9.9, I was at 31C or so.

Here's a screenshot of my dashboard:

[image]

and a screenshot of the system log looping forever:

[image]

I have about 12 or so devices in my network.  06m as controller and 06 as a router.  Both are running 2.0.6-dev.

Rebooting the 06m seems to fix the issue for a few hours.

Krazy998, 2023-11-17:
> I have about 12 or so devices in my network. 06m as controller and 06 as a router. Both are running 2.0.6-dev.
> 
> Rebooting the 06m seems to fix the issue for a few hours.

I think this issue is separate to the temperature issue which this is related to. I would suggest maybe creating a new issue which focus's on the "No space for new connection issue"?

jonnycastaway, 2023-11-17:
@Krazy998 
At my side, 2.0.7 solves the speed issue with zigbee reaction.
My sensor, specially my PIRs are react now like in 0.9.9.
What's on your's? Also ok?

Krazy998, 2023-11-17:
> 2.0.7 solves the speed issue

Yes it's now perfect. Very fast response times from all zigbee devices

jonnycastaway, 2023-11-17:
Ok, sounds very good.
Now back to the original topic :-)
The temperatures are also high in 2.0.7
I'm curious what't the bug, the temp or the unit.

xvisor16, 2023-11-18:
Although I didn't experience any slowness of my Zigbee netwerk with 2.0.6. I did experience the webinterface of the slzb-06 being very slow, with 2.0.7, the webinterface is lightning fast again. Zigbee still working perfect. Temperature remained the same here 47/50 celsius. The casing of the device feels 10-15 degrees colder at least on the hottest spot.

smlight-dev-223942, 2023-12-10:
> Although I didn't experience any slowness of my Zigbee netwerk with 2.0.6. I did experience the webinterface of the slzb-06 being very slow, with 2.0.7, the webinterface is lightning fast again. [...]

can you please try 2.0.8 version at least and check?

smlight-dev-223942, 2023-12-10:
> Hello, I'm running 2.0.6-dev on my 06m and I noticed that temperatures were hotter than normal. I'm on PoE - on 0.9.9, I was at 31C or so.
> 
> Here's a screenshot of my dashboard:

> I have about 12 or so devices in my network. 06m as controller and 06 as a router. Both are running 2.0.6-dev.
> 
> Rebooting the 06m seems to fix the issue for a few hours.

releases up to 2.0.8 were with some bugs (and anyway, they are in test mode as fw is based on completely new architecture), but starting from 2.0.8 it is more or less reliable :) By the way, 0.9.9 so far is rock solid

sikousikou, 2023-12-12:
[image]
 No Wifi,No Poe

DerDreschner, 2023-12-12:
@sikousikou How does the environment look like where the SLZB-06 is mounted? Could you show a picture?

sikousikou, 2023-12-12:
[image]

I have a temperature/humidity sensor that is located less than 30 cm next to the Zigbee module, here are the values

BlastyCZ, 2023-12-12:
> ![Screenshot_20231212_142431](https://private-user-images.githubusercontent.com/7949811/289873271-225d57a1-f0ac-4558-baa4-3cc5b5fb92fa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIzOTE4MjUsIm5iZiI6MTcwMjM5MTUyNSwicGF0aCI6Ii83OTQ5ODExLzI4OTg3MzI3MS0yMjVkNTdhMS1mMGFjLTQ1NTgtYmFhNC0zY2M1YjVmYjkyZmEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMTJUMTQzMjA1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDU3ZjM3NTA5MDQ3MDRhYmRlNzAxMGU4ZTdmOWM0OTBhZmMyNzExYjc2Zjc4N2Y2MGFkMjg3ZmQyY2Q0NjFlMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.N4p2RcPUl8IJ7thV3MSGzTfmRgDhq63d98Eu1cGox9k) No Wifi,No Poe

My temperatures are even worse, same setup