smlight-dev/slzb-06-firmware #74: Feature Request: WiFi Fallback

villamanuel1980, 2023-11-15:
Hello everyone!

We're using these antennas profesionally. Sometimes an antenna will lose lan connectivity and we will lose any way to restart the ESP32 it or see what's going on.

It would be very useful for us to have a feature that let's the antenna connect to an available WiFi if LAN connectivity is lost, as a fallback.

Hope this can be considered in future versions. Keep up the good work!

Thanks!

smlight-dev-223942, 2023-12-10:
can you tell us more? the feature request looks very good, although it is not clear why your device lost connectivity.

gluoncompany, 2023-12-11:
In our case, the LAN connectivity was lost due to a faulty ethernet port in the coordinator (as far as I can tell).

The coordinator detected no connectivity over ethernet, but there was a WiFi available to it. It would be great if we could tell the coordinator that in case ethernet is lost (faulty switch or whatever else), it can try and connect to WiFi. 

We're remote to the coordinators, and are using uptime Kuma on the Z2M to watch their health, but we're lacking some other tools maybe in the coordinators themselves to monitor them in the long term.

I'm happy to give any details you need to test this.

Thanks!