smlight-dev/slzb-06-firmware #77: How to flash to v.2.x using online flasher?

quantummobile, 2023-11-22:
Hey there,

I can't find any meaningful information anywhere on how to actually flash to v.2.x from v.0.9.9 using online flasher?
[image]
It only allows to flash to the same v 0.9.9 or lastes from 2.x branch, which I guess is wrong?

Very much look forward to hearing from you for support on this one.

Cheers

Krazy998, 2023-11-22:
Just flash 2.0.8 it works fine

neuhausf, 2023-11-24:
What are the changes in 2.0.8? I don't see any changelog.

alexsalvo1, 2023-11-25:
You need a data cable otherwise it doesn’t appear as a device

Coldness00, 2023-11-25:
Hello,
I use z2m and I would like to update to this new firmware.
Do I have to re pair all my devices afterward?
Thank you in advance.

Arn0uDz, 2023-11-25:
> Hello, I use z2m and I would like to update to this new firmware. Do I have to re pair all my devices afterward? Thank you in advance.

I went from 0.9.9. to this and no need to re pair.

closed 2023-12-10 by smlight-dev-223942