smlight-dev/slzb-06-firmware #78: Increase the number of objects per key - router

jerome6994, 2023-11-28:
Good morning
I have a little over 200 objects (205 exactly) and the key seems to be indicated for 150 devices maximum.
Is it possible to push the number of objects beyond please.
In fact, I cannot set up several Zigbee networks because my home is made of machefer and the mesh is complicated and requires a lot of routers to work well.
Today I'm getting by with a Conbee II but I would have liked to replace it with your POE Zigbee key or even POE Zigbee + Matter
Thanks

smlight-dev-223942, 2023-12-10:
sorry for late reply. It also depends how many routers do you have in your set up. can you tell me? Theoretically, it can work with 205 devices. We can test, write to our email so we can organize that test

closed 2023-12-10 by smlight-dev-223942

jerome6994, 2023-12-11:
Good morning
No worries about the late response, a late response is better than no response at all 

Otherwise I did a little cleaning up my Zigbee network and I only have 193 devices left but that changes from time to time.

I now have 193 devices including 81 routers and 112 end devices on zigbee2mqtt

I'm looking for your email and writing to you

jerome6994, 2023-12-11:
Good morning
No I haven't solved this problem.
So I'm still on the conbee 2 key and I haven't switched to the SLZB-06M key

Thank you for your help and/or extension number of devices per key.