smlight-dev/slzb-06-firmware #81: Firmware source code?

hakong, 2023-12-09:
Hi,
I can't seem to find the source code for the 2.0 firmware - where is it?
The 0.9.9 version is "deprecated" and flashing the 2.x firmware is recommended but there are no 2.x releases, yet the online flasher offers it.

smlight-dev-223942, 2023-12-10:
Replied here https://github.com/smlight-dev/slzb-06-firmware/issues/80#issuecomment-1849108224 

By the way, FW is not "recommended - it is just different because it is built on our new architecture for all SLZB/SLWF devices family. It is not the successor of 0.9. Both versions work (to be honest, 0.9.9 is even more stable so far, as some users still report bugs in 2.x). Feel free to email us if you have any more questions.

closed 2023-12-10 by smlight-dev-223942

hakong, 2023-12-11:
[image]

Related to https://github.com/smlight-dev/smlight-uniflasher-firmwares/issues/3, I am quite sceptical of all the firmware/software development/release/flashing around this product. I should have done more research before buying it.

**I would not have bought this if I had discovered that before.**

The development[1] is not very transparent[2], the firmware links lead to YouTube videos[3], you recommend flashing firmware blobs downloaded without any transparent development and build process[4], the firmware being flashed by the online flasher is not on GitHub[5].

I would recommend uploading the source code for 2.x ASAP, develop it transparently and build it on GitHub if possible.

[1] Who is smlight-dev-223942?
[2] https://github.com/smlight-dev/smlight-uniflasher-firmwares/commit/a9df8c1b0495ddb96be901b289bd3c5d8d6fe847
[3] [Latest FW: [v2.x](https://youtu.be/q3cWHQ8CdcA)](https://github.com/smlight-dev/slzb-06-firmware?tab=readme-ov-file#latest-fw-v2x)
[4] https://github.com/smlight-dev/smlight-uniflasher-firmwares/tree/main/firmwares
[5] https://smlight.tech/flasher/#SLZB-06M flashes 2.0.11-dev but [4] shows the latest being 2.0.1 for slzb-06m and 2.0.7 for slzb-06

BoBeRzE, 2023-12-11:
I would also like to buy the SLZB-06M. There is advertising for open source firmware but the current firmware is not publicly available. For this reason I will not buy the device. That is not very trustworthy

Tarik2142, 2023-12-11:
> ![image](https://private-user-images.githubusercontent.com/4422113/289487870-95033f4a-5a69-4335-982c-ef61398eeb14.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIyOTk5NzksIm5iZiI6MTcwMjI5OTY3OSwicGF0aCI6Ii80NDIyMTEzLzI4OTQ4Nzg3MC05NTAzM2Y0YS01YTY5LTQzMzUtOTgyYy1lZjYxMzk4ZWViMTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIxMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMTFUMTMwMTE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2EyOGE5M2MzODI2NzI2MDc0ZGM2MThkN2M0ZjcxNTE3MzJjNDVkYzlmNzlmZjA0MDc4MzhmZjg2NmQ1YTc1NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.FaYrYbNbyxWqTU8f-hMhPrl_n-TdBdJaeKIvnN9fkTY)
> 
> Related to [smlight-dev/smlight-uniflasher-firmwares#3](https://github.com/smlight-dev/smlight-uniflasher-firmwares/issues/3), I am quite sceptical of all the firmware/software development/release/flashing around this product. I should have done more research before buying it.
> 
> **I would not have bought this if I had discovered that before.**
> 
> The development[1] is not very transparent[2], the firmware links lead to YouTube videos[3], you recommend flashing firmware blobs downloaded without any transparent development and build process[4], the firmware being flashed by the online flasher is not on GitHub[5].
> 
> I would recommend uploading the source code for 2.x ASAP, develop it transparently and build it on GitHub if possible.
> 
> [1] Who is smlight-dev-223942? [2] [smlight-dev/smlight-uniflasher-firmwares@a9df8c1](https://github.com/smlight-dev/smlight-uniflasher-firmwares/commit/a9df8c1b0495ddb96be901b289bd3c5d8d6fe847) [3] [Latest FW: [v2.x](https://youtu.be/q3cWHQ8CdcA)](https://github.com/smlight-dev/slzb-06-firmware?tab=readme-ov-file#latest-fw-v2x) [4] https://github.com/smlight-dev/smlight-uniflasher-firmwares/tree/main/firmwares [5] https://smlight.tech/flasher/#SLZB-06M flashes 2.0.11-dev but [4] shows the latest being 2.0.1 for slzb-06m and 2.0.7 for slzb-06

Firmwares are transferred to our server.
Firmware 2.0 was developed from scratch and if a decision is made to publish it, you will know about it.
The problem with open source is that some people copy your device and take your open firmware and here is a complete copy of your device in which you invested a lot of effort and time. I don't think you would want that to happen to you either.

Krazy998, 2023-12-11:
If you are worried source code, you could most likely flash your own by using esphome? 

Discribed here: https://github.com/smlight-dev/slzb-06-esphome/blob/main/slzb-06-esphome-zigbee.yaml