smlight-dev/slzb-06-firmware #82: no web interface after upgrade to 2.0.8-dev

timota, 2023-12-10:
After upgrade via web flasher from 0.9.9 to 2.0.8-dev no web interface.

I can see that port 80 is opened(nmap) but no response at all. 

No options to upgrade to previous firmware versions even on this repo it recommended to use FW 2.xx

FedericoAntoniazzi, 2023-12-10:
Hi @timota, I tried the same firmware version but I cannot connect to my SLZB-06M by using WiFi.
As far I can see, in my case the device is unable to acquire an IP address but works perfectly in Ethernet mode.

Anyway, you can download older binaries from the [uniflasher repo](https://github.com/smlight-dev/slzb-06-firmware) and do a manual flash, which can be achieved with esptool.py (https://github.com/espressif/esptool)

smlight-dev-223942, 2023-12-10:
> After upgrade via web flasher from 0.9.9 to 2.0.8-dev no web interface.
> 
> I can see that port 80 is opened(nmap) but no response at all.
> 
> No options to upgrade to previous firmware versions even on this repo it recommended using FW 2.xx

Hi timota, you can flash to your initial fw [here](https://smlight.tech/flasher/#SLZB-06). fw 2 is new, that is why some bugs can appear. By the way, it will help us to analyze - after flashing back, can you please send to our email a screenshot of your dashboard from the previous firmware?

closed 2023-12-10 by smlight-dev-223942